khoih-prog/RP2040_PWM #2: Change the PWM frequency

americodias, 2021-10-04:
Hi Khoi,

### Describe the bug

I'm not sure if this is a bug or a "feature", but I wanted to dynamically change the PWM frequency and I'm not able to do so using the setPWM function.

### Steps to Reproduce

Example code: https://pastebin.com/raw/HXSsr0yr

### Expected behavior

I was expecting to have 2s of 1000Hz and then 2s of 2000Hz repeatedly.

### Actual behavior

Frequency stays constant at 1000Hz.

Thanks,
Américo

americodias, 2021-10-04:
Hi again,

I forgot to mention that if I instantiate new objects each time it works:

```
void loop()
{
  delay(200);
  frequency = 1000;
  PWM_Instance = new RP2040_PWM(pin0, frequency, 50);
  PWM_Instance->setPWM();

  delay(200);
  frequency = 2000;
  PWM_Instance = new RP2040_PWM(pin0, frequency, 50);
  PWM_Instance->setPWM();

}
```

But this is not optimal I guess.

Kind regards,
Américo

khoih-prog, 2021-10-04:
Hi @americodias 

Thanks for reporting an interesting bug due to my fault. I'm releasing by today the new release v1.0.2 to fix the issue.

I'd really appreciate your contribution. Happy bug finding.

Best Regards,

KH

khoih-prog, 2021-10-04:
Hi @americodias 

The new [**RP2040_PW releases v1.0.2**](https://github.com/khoih-prog/RP2040_PWM/releases/tag/v1.0.2) has just been published

Your contribution again has been noted in [**Contributions and Thanks**](https://github.com/khoih-prog/RP2040_PWM#contributions-and-thanks)

The library is definitely better and more useful now. Looking forward to receiving more bug reports from you.

Best Regards,

KH

---

### Releases v1.0.2

1. Fix bug not changing frequency dynamically. See [Change the PWM frequency #2](https://github.com/khoih-prog/RP2040_PWM/issues/2)

closed 2021-10-04 by khoih-prog

americodias, 2021-10-05:
Hi @khoih-prog,

Thanks for the quick fix. I will test it asap.

Kind regards,
Américo

khoih-prog, 2021-10-05:
Hi,

Be sure to test using the new and better [**RP2040_PWM** releases v1.0.3](https://github.com/khoih-prog/RP2040_PWM/releases/tag/v1.0.3)

Thanks,

---

### Releases v1.0.3

1. Not reprogram if same PWM frequency. 
2. Add PIO strict `lib_compat_mode`