khoih-prog/RP2040_PWM #3: Attempting to Alter the Duty Cycle w/o changing any other values

AKLitman, 2022-01-06:
Arduino IDE 1.8.19
RP2040_PWM - v1.0.4
Using Example - PWM_DynamicFreq

I'd like to modify the Duty Cycle w/o altering the Frequency.  My goal is to change the speed of a DC Motor
in a small project.  When I change the Duty Cycle in the example without changing the frequency, the Duty Cycle
remains the same as it was before the "SetPWM" call.  I looked at your code to see if I was using the wrong call
but do not see a specific call for altering only the Duty Cycle.  If I alter the Frequency in your example by any
amount, the Duty Cycle is changed as I specify in the SetPWM call.  If I do not alter the Frequency, the Duty Cycle
remains unchanged.

Can you show me the proper way to alter the Duty Cycle w/o changing any other values?

khoih-prog, 2022-01-06:
Hi @AKLitman 

Thanks for opening the issue which I can consider as a real bug or an enhancement request.

I'll fix and create a new release soon to let you alter the DC on-the-fly.

khoih-prog, 2022-01-07:
Hi @AKLitman 

The new [**RP2040_PWM** releases v1.0.5](https://github.com/khoih-prog/RP2040_PWM/releases/tag/v1.0.5) has just been published. Your contribution is noted in [Contributions and Thanks](https://github.com/khoih-prog/RP2040_PWM#contributions-and-thanks)

Please try the new example [**PWM_DynamicDutyCycle**](https://github.com/khoih-prog/RP2040_PWM/tree/main/examples/PWM_DynamicDutyCycle) which demonstrate how to change dutyCycle and keep same frequency on-the-fly.

Best Regards,

---

### Releases v1.0.5

1. Permit changing dutyCycle and keep same frequency on-the-fly. Check [Attempting to Alter the Duty Cycle w/o changing any other values #3](https://github.com/khoih-prog/RP2040_PWM/issues/3)
2. Add example [PWM_DynamicDutyCycle](https://github.com/khoih-prog/RP2040_PWM/tree/main/examples/PWM_DynamicDutyCycle) to demonstrate new feature
3. Add support for more boards using arduino-pico core, such as ADAFRUIT_ITSYBITSY_RP2040, CYTRON_MAKER_NANO_RP2040, SPARKFUN_PROMICRO_RP2040, CHALLENGER_2040_WIFI_RP2040, ILABS_2040_RPICO32_RP2040, MELOPERO_SHAKE_RP2040, SOLDERPARTY_RP2040_STAMP, UPESY_RP2040_DEVKIT, WIZNET_5100S_EVB_PICO, etc.
4. Display compiler informational warning messages only when `_PWM_LOGLEVEL_` > 3

closed 2022-01-07 by khoih-prog

AKLitman, 2022-01-07:
Thank you.  I will download and try this out.  Your code is very straightforward and I found the library easy to install and use.  This new feature will greatly enhance its use for speed control on the Cytron Maker Pi board.  Austin

AKLitman, 2022-01-07:
UPDATE:  The new Feature works perfectly.  My Cytron Maker Pi board responds exactly as intended now.  Many thanks for being very responsive and accommodating.  Austin