khoih-prog/RP2040_PWM #4: Setting duty cycle without disturbing frequency

Gavin-Perry, 2022-02-20:
### Describe the bug
Added in 1.0.5: 
 1. Permit changing dutyCycle and keep same frequency on-the-fly should now work but it's not clear how.
 2. Add example - Example is not clear and is not included in the downloaded library from Arduino
 3. Not sure why bards need to be listed but I'm using the WaveShare RP2040-Zero  $5 and small and castellated, just what I need for including in my project.  Only difference from a generic is the cute little WS2812 RGB LED which I can't get to work.
 4.  Yep, I can lower the debug level, though I'm not sure what those warnings are about.
 5. 
A clear and concise description of what the bug is.
What are the units you are using for duty cycle? Normally with 16 but PWM you would just load a 16 bit count value and be done with it. It looks like you might be using float of 0.0 - 100.0% - Trying to be fast here why mess with floating point numbers?
I tried both 
    PWM_Instance[0]->setPWM(pin0, freq, (double(lookup8to16[i]), false);  //  Pass 16 bit value, but as a double (which is just a float?)
    PWM_Instance[2]->setPWM(pin2, freq, (double(lookup8to16[(i])/655.35, false);  // 
  // lookup8to16[] is a 256 value array of the gamma curve so brightness is properly log scale for human eye
 

### Steps to Reproduce
Used your example PWM_Multi adding to the loop:
  for (uint i=255; i>0; i-- ) {
     PWM_Instance[0]->setPWM(pin0, freq, (double(lookup8to16[i]), false);  // 
    PWM_Instance[2]->setPWM(pin2, freq, (double(lookup8to16[(i+0x80)&0xff])/655.35, false);  // offset this one by 1/2 the table
    delay(30);
  } 

Also tried
// Do it by reaching in past the library to the PWM hardware
    pwm_set_counter (pwm_gpio_to_slice_num(pin0), (lookup8to16[i]/655.35));  // 0 to 100%?
   pwm_set_counter (pwm_gpio_to_slice_num(pin1), lookup8to16[(i+64)&0xff]); // 16 bit value?
   
### Expected behavior
I'm looking for a smooth increase and then decrease in the duty cycle (and thus LED brightness) from 0 to 100% and back to 0

### Actual behavior
Depending on which version I try I get some smooth action then big jumps in brightness (something is wrapping to zero at the wrong time). Even when it is working over some of the range it's flickering when it shouldn't (resetting the slice? IDK)

### debug
set level to 3 don't get any reports
One odd thing is that when I enable the Serial Monitor it always says (in orange at the top of Monitor pane):
"Not Connected. Select a board and port to connect automatically"
But then it works anyway.  Took me a while to realize it's in a pane of the program window not a separate window as in 1.8
1.8 couldn't find the PIO.h  file

### Screenshots
I could send a video of the LED or scope traces of the PWM (Have to set up the o'scope)

### Information

* Arduino IDE 2.0.0 rc3 (also tried 1.8.15 but messed that up installing arduino-pico)
* `RP2040-Zero from WaveShare 
* Contextual information 
  Smooth deep dimming down to 1/65536 full 16 bit dimming (most commercial dimmers are 10%-100%  and "best" e.g. DALI are still only 1:1000 i.e. 0.1% Eyes can see much better than that.

* Simplest possible steps to reproduce
Using PWM_Multi but removed array for frequence fixed freq = 2000; 
Loop changed to above code

* Anything that might be relevant in your opinion, such as:
 Aspire 5 core i5 2.4GHz 20GB Ram; Windows 11 Home 21H2  - if that matters
  I'd like a simple to use library, the whole point of a library is to hide complexity so once the PMW slice has been set up there should be simple routines like
  setPWMfreq(pin#,newfreq);
  setPWMdcPct(pin#,newdc); // In percent if people really want that
  setPWMdc16(pin#, uint newDCval);  // 16 bit value

I've been reading the Pico hardware manual, it's dense, this machine is so powerful!
I'm happy that you dug in and extracted a working PWM package, it just needs a little tweeking and documentation I think.

BTW another odd thing:
After working for quite a while loading code after compile just fine. It quit loading and now requires unplugging and holding boot button every time. Is that because I'm using all the PIO?

khoih-prog, 2022-02-21:
HI @Gavin-Perry 

You have to write a better and concise issue report. I won't spend time to go thru your rant, without enough and correct info to test and duplicated.

Please follow and post the standard [**MRE**](https://stackoverflow.com/help/minimal-reproducible-example) so that anyone can duplicate without wasting time.

Also post the issue to the correct library or core. Separate the issue to many issues as necessary.

I have to close this and wait for anything better to act upon.

closed 2022-02-21 by khoih-prog

Gavin-Perry, 2022-02-21:
Khoi,

I apologize for the issue post on duty cycle appearing to be a rant. I
posted it after a long frustrating day well after midnight and really
should have read it over the next day before posting it. I thought I was
covering all the requirements of an issue as  I followed the outline given
in the issue template.
Ill check the MRE link in the future and be sure not to mix other issues
into the post (especially since the git issue space is a public forum.

Upon rereading your Multi example I see that you are in fact using % for
duty cycle.
The issue is converting from double to uint16_t  seems to be affecting the
smoothness.
I'll work on it some more and create a concise example that you can run.

THe second part is a suggestion that there be a simple SetPWMDC16bit
routine (or whatever you would want to call it so that I can drop in my 16
bit duty cycle instead of converting to float and back.

Thanks again for the support and clarification about MRE. I owe you another
cup of coffee. ;-)

Gavin

khoih-prog, 2022-02-21:
Hi @Gavin-Perry 

I'm sorry if you feel offended. I really didn't mean so.

We're all in the hi-tech, not sales and used to speak not in diplomatic way to have fast and good results.

I'm inviting you into a private repo, where many experts, I handpicked, have been participating and we can discuss / support each other relating to many more hi-tech subjects, sometimes more personally and suitable than in this public repo.

Regards,

**PS**: Please don't buy me anymore coffee as you've bought enough. It's so symbolic for me to find a reason to continue this library writing and supporting for free. Just to see if how many people are really caring and thankful for the work.

I've received some time ago some **ungrateful** responses and almost stopped doing the work.

Just try to continue, to keep my brain from idling, until not possible or bearable anymore. 

I'm so good financially, have been early-retiree for nearly 20 years (after being co-owner / tech executive for a hi-tech Silicon Valley company), and have spent a lot of precious time as well as good fortune to buy almost every boards / system / test equipment to provide support, which had been lacking in the field. 

Certainly very far from `Bill Gates`, who, people jokingly said, refuses to pick up the 100 bucks as it costs more to do so ;-))

Gavin-Perry, 2022-02-22:
Hi Khoi,

I was just afraid I had offended you.  I am surprised to learn how similar
we are!

Here is a little background on me:

I've been in the rough and tumble world of software development since the
1970's when everyone was polite and full of inside jokes (foo.bar mung etc)
flaming (same as ranting now?) existed but was frowned upon. I was "the
computer guru" at Washington University in St. Louis for the 80s and into
the 90s mostly doing mini (PDP11) and micro (6502 and on up the ladder)
I got back into hardware for a while running the electronics shop until
2010.
The younger generation is so easily offended (and I am so good at being
offensive when I'm not careful, as my wife constantly reminds me :-)  ).
that I was worried.
When accused of ranting I have to look at myself long and hard, it's not
exactly a badge of honor.

In the programming world I've been out of touch for quite a while. I used
so many languages and IDEs over the decades it would make your head
spin.... Nearly everything from 6502 assembler to Pascal, C++, Forth...
just NOT COBOL. I wrote a ton of firmware back when I had to build my own
eprom burmer and other peripherals for the labs. (A/D D/A timers motor
controls etc)

I retired from the university in 2010 to do stuff that is more green.
Started my own company GPTS,LLC
Green Power Technology Solutions, or Good Practical Tools for Science I
thought.
My friends suggested:  Gavin Perry Tall Scientist
I've abandoned the web site (lost the password!) and didn't really need
more work, but it is still there
HTTP://GPTSLLC.COM  (Never secured it) Have a look at some of last decade's
projects.
I've been working part time for an LED importing company. I'm the entire
engineering department so no one to talk to there.  They just want stuff
cheaper so they can keep selling to WalFart.

Recently I decided to get back into  micro programming, but I don't learn
as fast as I used to so I'm sticking with Arduino. I've had git. access on
and off for quite a while, and I think it's sort of like subversion but I
was never great with those kind of tools having mostly been a one man shop.
I get the basics but have never done so much as a pull request.

I'm so grateful to be invited to your private repo. St Louis has an IT
community and hacker community but I (like everyone) have been isolated
the last couple of years. I'm happy to discuss projects and whatever and to
do it in a more private space where I can be as joking and offensive as the
others will tolerate.
(are you still reading?)

MEANWHILE:
I took a look at the RP2040 documentation (I just loved reading the docs
back in the day RTFM)
and realized I can do the PWM I want with what's in the standard PWM.h

So I was working on this when I got the notice that you had emailed me.
(Not finished or tested yet)

/***********************************************************************************
|   RawPicoPWM.ino
|   Control the PWM of a pin on a RP2040 (pico) with the built in functions
|   Can make this a library by wrapping "setup" into initPWM(chans,...)
|   pwm_set_gpio_level(gpio, x) seems clear enough to me to just use it.
|   Built by Gavin Perry (thanks to Khoi Hoang for getting me started)
|   Licensed under MIT license
|**********************************************************************************/

[... 69 lines cut ...]
    Serial.print("\t");
    Serial.println(lookup8to16[(i+0x80)&0xff]/655.35);
  }
    for (uint i=255; i>0; i-- ) {
// Do it by reaching in past the library to the hardware
// The library uses % 0-100
 // pwm_set_counter (pwm_gpio_to_slice_num(pin1),
lookup8to16[(i+64)&0xff]); // 16 bit value?
    PWM_Instance[0]->setPWM(pin0, freq, double(lookup8to16[i]/655.35), false
);  //
    PWM_Instance[2]->setPWM(pin2, freq, double(lookup8to16[(i+0x80)&0xff]/
655.35), false);  // offset this one by 1/2 the table
    pwm_set_gpio_level(pin1, ( _PWM_config.top * _dutycycle ) / 100 );
    delay(20);
    Serial.print("DMX val ");
    Serial.print(i);
    Serial.print("\t DC%");
    Serial.println(lookup8to16[i]/655.35);
//    Serial.print("Actual PWM DC "); Serial.println(PWM_Instance[0]->
getActualdutycycle()); /// oops you don't have a get count
//    Serial.print("Actual PWM DC "); Serial.println(PWM_Instance[2]->
getActualdutycycle());
  }
Anyway, good to meet you and I'll join your group now (assuming I can
figure it out)