khoih-prog/RP2040_PWM #5: Request for Clarification on PWM Slices and A/B sides

AKLitman, 2022-02-24:
Thanks for this library and for providing a Duty Cycle change function.  I am using version 1.12.0 now and tried creating 2 differing PWM pins on adjacent GPIO's (GP10 & GP11).  According to the Programmer's Model, GP10 would use 5A and GP11 would use 5B so I assumed they could have differing Duty Cycles with the same Frequency.  Alas, when I set GP10, GP11 stops and when I set GP11, GP10 stops.  I changed to adjacent pins GP11 and GP12 which use 5B and 6A which now works.  The board I am building controls two DC motors on a dual drive robot chassis.  Because my board is custom, I have the flexibility to manage my own pin assignments.  This is not the case with some commercial boards which have pre-assigned pin usage.  My question is, am I correct in that dual duty cycle pins have to be on non-adjacent slices or is there a special way to generate differing duty cycles on adjacent pins using the same slice but differing A & B sides?  Thank you - Austin

khoih-prog, 2022-02-24:
Hi @AKLitman 

This is actually a **limitation / bug** of this library, which currently not permits to use both channels (PWM_CHAN_A, PWM_CHAN_B) of the same slice to output PWMs (same freq. different duty-cycles)

Thanks to your report, I'm now aware, figure out and rewrite the library to permits to use both channels of the same slice.

I'll post a new release soon as I'm currently testing and it's working OK so far now.

I'm glad that the library will be **better and better** thanks to `contributing users` like you.

Thanks and Regards,

khoih-prog, 2022-02-25:
Hi @AKLitman 

The new [**RP2040_PWM** releases v1.1.0](https://github.com/khoih-prog/RP2040_PWM/releases/tag/v1.1.0) has just been published. Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/RP2040_PWM#contributions-and-thanks)

Please try the new example [**PWM_MultiChannel**](https://github.com/khoih-prog/RP2040_PWM/tree/main/examples/PWM_MultiChannel) which demonstrates how to use both channels of PWM slice.

Best Regards,

---

### Releases v1.1.0

1. Permit PWM output for both channels of PWM slice. Check [**Request for Clarification on PWM Slices and A/B sides** #5](https://github.com/khoih-prog/RP2040_PWM/issues/5)
2. Use float `instead` of `double` for frequency and duty-cycle
3. Add example [**PWM_MultiChannel**](https://github.com/khoih-prog/RP2040_PWM/tree/main/examples/PWM_MultiChannel) to demonstrate how to use both channels of PWM slice.

closed 2022-02-25 by khoih-prog