khoih-prog/RP2040_PWM #15: PWM_Basic example fails to blink LED on Pico W

badger200, 2023-01-06:
The Pico W's cy43 wireless chip takes over some of the GPIO and the LED and needs special handling.

I just tried this example and the LED doesn't blink.

khoih-prog, 2023-01-07:
HI @badger200 

The example is only for general RP2040. While RP2040W is very different, by using LED @ symbolic pin GPIO32, which is controlled only by CYW43 chip

https://github.com/earlephilhower/arduino-pico/blob/master/variants/rpipicow/pins_arduino.h#L8

https://github.com/earlephilhower/arduino-pico/blob/aeb41f3e70628e63ed1cf13f2031cdaf61d65ad0/variants/rpipicow/picow_digital.cpp#L35-L49

Therefore, the PWM can't control the RP2040W LED. 

Try to use a different pin and connect LED to it.

closed 2023-01-07 by khoih-prog

badger200, 2023-01-07:
I did find a reference example from official Raspberry Pi GitHub: https://github.com/raspberrypi/pico-examples/blob/master/pico_w/blink/picow_blink.c

And it works on my Pico W, however I did not try PWM fading.

khoih-prog, 2023-01-08:
You really don't need that fancy code for `RP2040W_BUILTIN_LED`

Just use normal code with `digitalWrite()`

```cpp
#define RP2040W_BUILTIN_LED       32

void setup() 
{
  pinMode(RP2040W_BUILTIN_LED, OUTPUT);
}

void loop() 
{
  digitalWrite(RP2040W_BUILTIN_LED, HIGH);
  delay(2000);
  digitalWrite(RP2040W_BUILTIN_LED, LOW);
  delay(2000);
}
```